elementzonline/SIM800_MQTT #5: Doesn't work if the Serial ports are changed

solankigithub, 2016-08-04:
Hi,
First of all, a wonderful job for creating such library.
Issue: When i look into the source, "Serial" (Hardware Serial) has been used to communicate with the SIM800 modem and mySerial has been used to print the output.

I am using MEGA, where i need to use "Serial1" to communicate with SIM800 and "Serial" to print the output. So i replaced the occurrence of "Serial" with "Serial1" and "mySerial" with "Serial".
I didn't change any other code.
The library is not working anymore.. Can you please help to make it little flexible with respect to choosing the serial ports.

nithinkurian, 2016-08-04:
You have to do replace  occurrence of "SerialEvent" with "SerialEvent1". I think this will solve your problem.

solankigithub, 2016-08-04:
Thanks Nithin,
I changed the occurrence of serialEvent() to serialEvent1().
It's still not working and giving the below response on the Serial port.

```
AT

OK
+++AT
ATE1

OK

OK
AT+CREG?

+CREG: 0,1

OK
AT+CIPMUX=0
AT+CIPMODE=1

OK

OK
AT+CGATT?

+CGATT: 0

OK

Call Ready
AT+CGATT=1

OK
AT+CIPSTATUS

OK

STATE: IP INITIAL

SMS Ready
AT+CSTT="AIRTELGPRS.COM"

OK
AT+CIPSTATUS

OK

STATE: IP START
AT+CIICR

OK
AT+CIPSTATUS

OK

STATE: IP GPRSACT
AT+CIFSR

106.218.239.147
AT+CIPSTATUS

OK

STATE: IP STATUS
AT+CIPSTART="TCP","broker.mqttdashboard.com","1883"

OK
AT+CIPSTATUS

OK

STATE: TCP CONNECTING

CONNECT
MQTT.TCP_Flag = True
2
2
Connect Acknowledgment
 
0
0
PING Response
 
0
0
PING Response

```

solankigithub, 2016-08-04:
This is my arduino main program
`
#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
String MQTT_HOST = "broker.mqttdashboard.com";
/*
   MQTT host address
*/
String MQTT_PORT = "1883";
/*
   MQTT port
*/
SoftwareSerial mySerial(10, 11); // RX, TX
/*
   Software Serial through which mqtt events log is printed at 9600 baud rate
*/
void GSM_MQTT::AutoConnect(void)
{
  /*
     Use this function if you want to use autoconnect(and auto reconnect) facility
     This function is called whenever TCP connection is established (or re-established).
     put your connect codes here.
  */
  connect("clientId-i8ZSXDfphz", 0, 0, "", "", 1, 0, 0, 0, "", "");
  /*    void connect(char *ClientIdentifier, char UserNameFlag, char PasswordFlag, char *UserName, char *Password, char CleanSession, char WillFlag, char WillQoS, char WillRetain, char *WillTopic, char *WillMessage);
          ClientIdentifier  :Is a string that uniquely identifies the client to the server.
                            :It must be unique across all clients connecting to a single server.(So it will be better for you to change that).
                            :It's length must be greater than 0 and less than 24
                            :Example "qwerty"
          UserNameFlag      :Indicates whether UserName is present
                            :Possible values (0,1)
                            :Default value 0 (Disabled)
          PasswordFlag      :Valid only when  UserNameFlag is 1, otherwise its value is disregarded.
                            :Indicates whether Password is present
                            :Possible values (0,1)
                            :Default value 0 (Disabled)
          UserName          :Mandatory when UserNameFlag is 1, otherwise its value is disregarded.
                            :The UserName corresponding to the user who is connecting, which can be used for authentication.
          Password          :alid only when  UserNameFlag and PasswordFlag are 1 , otherwise its value is disregarded.
                            :The password corresponding to the user who is connecting, which can be used for authentication.
          CleanSession      :If not set (0), then the server must store the subscriptions of the client after it disconnects.
                            :If set (1), then the server must discard any previously maintained information about the client and treat the connection as "clean".
                            :Possible values (0,1)
                            :Default value 1
          WillFlag          :This flag determines whether a WillMessage published on behalf of the client when client is disconnected involuntarily.
                            :If the WillFlag is set, the WillQoS, WillRetain, WillTopic, WilMessage fields are valid.
                            :Possible values (0,1)
                            :Default value 0 (Disables will Message)
          WillQoS           :Valid only when  WillFlag is 1, otherwise its value is disregarded.
                            :Determines the QoS level of WillMessage
                            :Possible values (0,1,2)
                            :Default value 0 (QoS 0)
          WillRetain        :Valid only when  WillFlag is 1, otherwise its value is disregarded.
                            :Determines whether the server should retain the Will message.
                            :Possible values (0,1)
                            :Default value 0
          WillTopic         :Mandatory when  WillFlag is 1, otherwise its value is disregarded.
                            :The Will Message will published to this topic (WillTopic) in case of involuntary client disconnection.
          WillMessage       :Mandatory when  WillFlag is 1, otherwise its value is disregarded.
                            :This message (WillMessage) will published to WillTopic in case of involuntary client disconnection.
  */
  
}
void GSM_MQTT::OnConnect(void)
{
  /*
     This function is called when mqqt connection is established.
     put your subscription publish codes here.
  */
  subscribe(0, _generateMessageID(), "sandeepsolanki/s", 2);
[... 62 lines cut ...]
     You can write your code here
  */

}

void loop()
{

  /*
     You can write your code here
  */
  if (MQTT.available())
  { 
    Serial.println("MQTT AVAILABLE");
    
    /*
      if you want to do something when mqtt connection is live.
      You can write your code here
    */
  }

  MQTT.processing();
}

`

solankigithub, 2016-08-04:
This is my GSM_MQTT.cpp
```

/*
  MQTT.h - Library for GSM MQTT Client.
  Created by Nithin K. Kurian, Dhanish Vijayan, Elementz Engineers Guild Pvt. Ltd, July 2, 2016.
  Released into the public domain.
*/

#include "GSM_MQTT.h"
#include "Arduino.h"
#include <SoftwareSerial.h>
#include <avr/pgmspace.h>
extern uint8_t GSM_Response;

extern SoftwareSerial mySerial;
extern String MQTT_HOST;
extern String MQTT_PORT;

extern GSM_MQTT MQTT;
uint8_t GSM_Response = 0;
unsigned long previousMillis = 0;
//char inputString[UART_BUFFER_LENGTH];         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete
void serialEvent1();
GSM_MQTT::GSM_MQTT(unsigned long KeepAlive)
{
  _KeepAliveTimeOut = KeepAlive;
}

void GSM_MQTT::begin(void)
{
  Serial.begin(2400);
  Serial1.begin(2400);
  Serial1.write("AT\r\n");
  delay(1000);
  _tcpInit();
}
char GSM_MQTT::_sendAT(char *command, unsigned long waitms)
{
  unsigned long PrevMillis = millis();
  strcpy(reply, "none");
  GSM_Response = 0;
  Serial1.write(command);
  unsigned long currentMillis = millis();
  //  Serial.println(PrevMillis);
  //  Serial.println(currentMillis);
  while ( (GSM_Response == 0) && ((currentMillis - PrevMillis) < waitms) )
  {
    //    delay(1);
    serialEvent1();
    currentMillis = millis();
  }
  return GSM_Response;
}
char GSM_MQTT::sendATreply(char *command, char *replystr, unsigned long waitms)
{
  strcpy(reply, replystr);
  unsigned long PrevMillis = millis();
  GSM_ReplyFlag = 0;
  Serial1.write(command);
  unsigned long currentMillis = millis();

  //  Serial.println(PrevMillis);
  //  Serial.println(currentMillis);
  while ( (GSM_ReplyFlag == 0) && ((currentMillis - PrevMillis) < waitms) )
  {
    //    delay(1);
    serialEvent1();
    currentMillis = millis();
[... 789 lines cut ...]
            Serial.print("Message ID :");
            Serial.println(MQTT.inputString[0] * 256 + MQTT.inputString[1]) ;
          }
          else if (ReceivedMessageType == PINGREQ)
          {
            MQTT.TCP_Flag = false;
            MQTT.pingFlag = false;
            Serial.println("Disconnecting");
            MQTT.sendATreply("AT+CIPSHUT\r\n", ".", 4000) ;
            MQTT.modemStatus = 0;
          }
        }
      }
      else if ((inChar = 13) || (inChar == 10))
      {
      }
      else
      {
        Serial.print("Received :Unknown Message Type :");
        Serial.println(inChar);
      }
    }
  }
}
```

solankigithub, 2016-08-08:
Hey Guys, any help ?
The publish is working when i call it inside the loop().
But unfortunately, the subscribe not working, I am sending messages but not able to receive it to my Arduino i.e the OnMessage event/method doesn't execute.

Thanks

guillebot, 2016-10-13:
Have you found any answers? I have the same problem. The library is working, but MQTT.available() returns allways false and some things behave weird.

durvesh09, 2016-11-23:
Same issue !! not working.. got stuck at MQTT.MessageFlag = true;
any solution ?

nictally, 2016-12-22:
I got it to work with Altsoft serial on pin 8 and 9

karanmakharia, 2017-01-04:
@nictally Can you tell me the changes you did in the cpp file and the example file. For me the output is just stuck at +++AT i.e. SerialEvents
@solankigithub Did you get it working?

arminsalcin, 2017-02-07:
Guys for me all works like a charm when i rewrite serial event. I will upload my code in some days :D

guillebot, 2017-02-07:
Can't wait to try it! Thanks.

arminsalcin, 2017-02-07:
Here it is  https://files.fm/f/6qwyj9z9

karanmakharia, 2017-02-07:
@arminsalcin Hello. I need to know if this code works even for software serial because I don't want to use Mega. I will be using nano, pro mini or Uno with a software serial attached to the Sim808 module.

arminsalcin, 2017-02-07:
I need to test it first with software serial , then if i make it it will be here.
Have a nice day :D

arminsalcin, 2017-02-07:
I found a way for Software serial but have a problems with SerialEvent() fuction becouse it work like a hardware interrupt when some informations come in serial it will run , so if we use a software serial nothing came in and code bug but if i connect arduino hardware serial RX with Software serial TX its run normaly. So found way but don't know is it okay to your project.

karanmakharia, 2017-02-15:
@arminsalcin If I wanted to connect it on the 0,1 pins that is Serial of arduino uno, then do I need to replace all the Serial1.write/Serial1.print functions with Serial.write/print resp? 
Currently the Serial is used for notifying us right?

abhaysbharadwaj, 2017-02-16:
Hello all,
I am getting this error:

```
C:\Users\Abhay.S.Bharadwaj\Desktop\sim900_mqtt_test\sim900_mqtt_test.ino: In function 'void loop()':

sim900_mqtt_test:52: error: cannot convert 'GSM_MQTT::available' from type 'bool (GSM_MQTT::)()' to type 'bool'

   if (MQTT.available)

                     ^

sim900_mqtt_test:58: error: '_generateMessageID' was not declared in this scope

     MQTT.publish(0, 1, 0, _generateMessageID(), publishTopic, gasPayload);

                                              ^

Using library SoftwareSerial at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial 
exit status 1
**cannot convert 'GSM_MQTT::available' from type 'bool (GSM_MQTT::)()' to type 'bool'**
```

** I have not done any changes to the library. My code is as follows:**

`#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
#define gasPin A1
String MQTT_HOST = "test.mosquitto.org";
String MQTT_PORT = "1883";
char* WillTopic = "*********";
char* WillMessage = "offline";
char* publishTopic = "*********";

SoftwareSerial mySerial(10, 11); // RX, TX
void GSM_MQTT::AutoConnect(void)
{

  connect("kumbh1", 0, 0, "username", "password", 1, 1, 1, 1, WillTopic, WillMessage);
}
void GSM_MQTT::OnConnect(void)
{
  subscribe(0, _generateMessageID(), "SampleTopic", 1);
  publish(0, 1, 0, _generateMessageID(), "SampleTopic", "Hello");
}
void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength)
{
  mySerial.println("New MQTT Message Reseived! Details: ");
  mySerial.println(TopicLength);
  mySerial.println(Topic);
  mySerial.println(MessageLength);
  mySerial.println(Message);
}
GSM_MQTT MQTT(20); //20 is the keepalive duration in seconds

void setup()
{
  // initialize mqtt:
  // GSM modem should be connected to Hardware Serial
  //  index =0;
  MQTT.begin();

  /*
     Write your setup related arduino code like:
     pinMode(gasPin, INPUT);
  */

}
void loop()
{

  /*
     You can write your maincode here. Example:
  */
  float x = analogRead(gasPin);
  char* gasPayload = f2s(x, 0); //f2s is our function to convert float values to string
  if (MQTT.available)
  {
    /*
      if you want to do something when mqtt connection is live.
      You can write your code here
    */
    MQTT.publish(0, 1, 0, _generateMessageID(), publishTopic, gasPayload);
  }

  MQTT.processing();
}

/* float to string
   f is the float to turn into a string
   p is the precision (number of decimals)
   return a string representation of the float.
*/
char *f2s(float f, int p)
{
  char * pBuff;                         // use to remember which part of the buffer to use for dtostrf
  const int iSize = 10;                 // number of buffers, one for each float before wrapping around
  static char sBuff[iSize][20];         // space for 20 characters including NULL terminator for each float
  static int iCount = 0;                // keep a tab of next place in sBuff to use
  pBuff = sBuff[iCount];                // use this buffer
  if (iCount >= iSize - 1)
  { // check for wrap
    iCount = 0;                         // if wrapping start again and reset
  }
  else
  {
    iCount++;                           // advance the counter
  }
  return dtostrf(f, 0, p, pBuff);       // call the library function
}
`
can anyone help me out?

arminsalcin, 2017-02-16:
in void loop change _generateMessageID() to example "armin" with quotes put your name or whatever tou want to be your message id

arminsalcin, 2017-02-16:
MQTT.publish(0, 1, 0, "armin", publishTopic, gasPayload);

abhaysbharadwaj, 2017-02-16:
ok. I shall try it out!
Thank you for your quick response!

arminsalcin, 2017-02-16:
You're welcome :D , try it and say if it works.

Akelectro, 2017-04-30:
@karanmakharia I am also stuck at +++AT. Can someone please help? I am using the same code given above.
@arminsalcin Can you please upload your code again? The link seems to be expired
Any help would be appreciated guys. Thanks

arminsalcin, 2017-05-01:
Can you provide us more information , like what module you use? , where do you connect your module rx and tx ? , how do you power your module ...

jalkestrup, 2017-05-19:
Hi @arminsalcin  , thanks for the responses. As @Akelectro asked it would be great if you could upload the edited library that allows for the use of software-serial interface with the Sim800 :-) Despite that it might have problems in relation to SerialEvent().

arminsalcin, 2017-05-19:
@jalkestrup  Hello guys sorry for link expired , i will uplaod new one ASAP !

bidyutper, 2017-05-27:
@arminsalcin link plz..lin provided is broken

priya2212, 2017-06-20:
@arminsalcin Please share the link!

shijinjose, 2017-08-01:
@nithinkurian @solankigithub 
Hi am using SIM800 with ATmega328p with seial on (2,3)

I have changed the software serrial and hardware serial and also renamed the serialEvent to serialEvent1. 
But am getting response upto AT+CIPSTART. after sending connect request, am not getting any response

Here is my serial terminal output. can u please help me to solve this issue..?

AT

OK
ATE1

OK
AT+CREG?

+CREG: 0,2

OK
AT+CREG?

+CREG: 0,1
-5þAT+CIPMUX=0

OK
AT+CIPMODE=1

OK
AT+CGATT?

+CGATT: 1
-5þAT+CIPSTATUS

OK

STATE: IP INITIAL
AT+CSTT="www","",""

OK
AT+CIPSTATUS

OK

STATE: IP START
AT+CIICR

OK
AT+CIPSTATUS

OK

STATE: IP GPRSACT
AT+CIFSR

10.244.81.80
AT+CIPSTATUS

OK

STATE: IP STATUS
AT+CIPSTART="TCP","61.16.154.28","1883"

OÿAT+CIPSTATUS

OK

STATE: TCP CONNECTING

CONNECT
MQTT.TCP_Flag = True

arminsalcin, 2017-08-13:
@shijinjose you cant use serialevent1 on arduino uno (atmega328) becouse it have only one hardware serial port and thats serial event . You can use only pins 0,1- arduino or 2,3- atmega328 directly.

coffeerr2004001, 2017-11-14:
hi, your link expired again. @arminsalcin

arminsalcin, 2017-11-14:
@coffeerr2004001 
[MQTT.zip](https://github.com/elementzonline/SIM800_MQTT/files/1470471/MQTT.zip)

abhaysbharadwaj, 2018-02-15:
Tested with "Arduino Mega" and things work flawlessly. Was able to get the publish part on Arduino UNO/Nano over software serial. but Subscribe never worked.

again, everything is working flawlessly in MEGA/Serial1.

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
Can you share your code ? , what code you used ?

abhaysbharadwaj, 2018-02-15:
[SIM800L_MQTT.zip](https://github.com/elementzonline/SIM800_MQTT/files/1727802/SIM800L_MQTT.zip)
file is same as that given by @arminsalcin

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
`if (MQTT.available())
  {

    MQTT.subscribe(0, 0, "/topic", 1);

  }`

Try to add subscribe in loop mode ifit works try also to put in "on connect" function play little bit with it !

abhaysbharadwaj, 2018-02-15:
[GSM_MQTT_Arduino_Subscribe_Publish.zip](https://github.com/elementzonline/SIM800_MQTT/files/1727847/GSM_MQTT_Arduino_Subscribe_Publish.zip)
This is the file. I have modified it to work on software serial of Arduino Uno. Only publish and LWT works. Subscribe is not working.(it requires hardware serial for event generation).

abhaysbharadwaj, 2018-02-15:
I have tried all this, It's not working.
Tried to do subscribe in loop, in setup, in MQTT.available(). Didn't work.

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
Can you also please share connection diagram arduino - sim module

abhaysbharadwaj, 2018-02-15:
SIM800L -- Arduino Uno/nano
Vcc - powered by external buck converter
Gnd - Gnd
Tx - SoftwareSerial Rx (D5)
Rx - SoftwareSerial Tx (D6)

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
Yes you can't get data from subscribe because you changed function to your internal function but it doesn't have any interrupt functions with hardware so when your data maybe come you cant read it , cause rx buffer only stays it don't interrupt any function to read .

abhaysbharadwaj, 2018-02-15:
oh, then I have to see how to work around this! But I haven't changed any functions. Only Software Serial --> Hardware and Hardware --> Software Serial.

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
Try this , connect jumper from D5 to arduino serial RX pin , and change in software function "myserialevent" back as default "serialevent" . You should get and subscribe but this is hack not solution...  . And yes you changed function ..
[image]

abhaysbharadwaj, 2018-02-15:
ah! I should try this out! regarding the function, I only renamed it. I didn't think it will affect anything.

basically changes all the "Serial" to "mySerial" and vice versa and worked it around a little.

Now, what you say should work. Connecting a jumper from SIM800L Tx to Arduino Uno Hardware Rxo get the data and generate an event. I'll have to try this.

arminsalcin, 2018-02-15:
yes you should  stay connection also to D5 as you set in code its your rx pin , you only interrupting RX pin this way :)

serialEvent() - looks as normal function but it has actually interrupt function so whenever soem data comes to rx pin it start this function changing name at any kind will corrupt it . 

https://www.arduino.cc/en/Tutorial/SerialEvent

abhaysbharadwaj, 2018-02-15:
now I get it!! "Serial" and "serial" made me assume it was a custom function! :P

arminsalcin, 2018-02-15:
@abhaysbharadwaj 
I'm glad you understand it now , have a nice day !

abhaysbharadwaj, 2018-02-15:
Thank you very much for the insight and help. You too have a great day.

guillerone, 2018-09-02:
Hi.
I tried the code from @arminsalcin and @abhaysbharadwaj but MWTT.AVAILABLE() never succeeds. Can anyone help me out? I ran out of ideas.

My GSM module allows to set the communicating PINs and I set them to 2(TX) and 5 (RX), and they are wired to the MEGA's 19(RX) and 18(TX) (2 to 19 and 5 to 18). The serial seems to spit all OK messages, but it still doesn't work, it goes into a loop where it just starts all over again.
Below is the console output. Any help will be deeply appreciated. Thank you!

AT

OK
+++AT
ATE1

OK
AT+CREG?

+CREG: 0,1

OK
AT+CIPMUX=0
AT+CIPMODE=1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIPSTATUS

OK

STATE: IP INITIAL
AT+CSTT="ac.vodafone.es","vodafone","vodafone"

OK
AT+CIPSTATUS

OK

STATE: IP START
AT+CIICR

OK
AT+CIPSTATUS

OK

STATE: IP GPRSACT
AT+CIFSR

77.209.197.103
AT+CIPSTATUS

OK

STATE: IP STATUS
AT+CIPSTART="TCP","io.adafruit.com","1883"

OK

CONNECT OK
AT+CIPSTATUS

OK

STATE: CONNECT OK
AT+CIPSHUT

SHUT OK
+++AT

OK
ATE1

OK
AT+CREG?

+CREG: 0,1

OK
AT+CIPMUX=0
AT+CIPMODE=1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIPSTATUS

OK

STATE: IP INITIAL
AT+CSTT="ac.vodafone.es","vodafone","vodafone"

guillerone, 2018-09-02:
Nevermind, I found out that if I add a small delay in SerialEvent1, it works...

```
void serialEvent1()
{
  while (Serial1.available())
  {
    delay(5); //Small delay to make it work
    char inChar = (char)Serial1.read();
```

ramonpasato19, 2019-02-07:
Hola alguien q me ayude con q pines se une un arduino mega y el sim900, para q funcione el codigo, en el monitor serie se queda en "++AT"

shadababe04, 2019-03-18:
Hi I change the Serial to mySerial n use the 2,3 as Rx Tx in Arduino Uno with SIM800, but still didn't get any response after sending connect request, am not getting any response Here is my serial terminal output. can u please help me to solve this issue..?

AT+CIPSTATUS

12:15:12.218 -> OK
12:15:12.218 -> 
12:15:12.218 -> STATE: IP INITIAL
12:15:12.265 -> AT+CSTT="WWW"

12:15:12.265 -> OK
AT+CIPSTATUS

12:15:12.343 -> OK
12:15:12.343 -> 
12:15:12.343 -> STATE: IP START
STATE: IP GPRSACT
12:15:29.406 -> AT+CIFSR

12:15:29.406 -> 100.115.165.214
12:15:29.453 -> AT+CIPSTATUS

12:15:29.499 -> OK
12:15:29.499 -> 
12:15:29.499 -> STATE: IP STATUS
12:15:29.499 -> AT+CIPSTART="TCP","broker.hivemq.com","1883"

12:15:29.640 -> OK

12:15:30.515 -> CONNECT
12:15:30.515 -> MQTT.TCP_Flag = True

shadababe04, 2019-03-19:
any help???

shadababe04, 2019-04-10:
> Same issue !! not working.. got stuck at MQTT.MessageFlag = true;
> any solution ?

i change the sim its working now

agozie, 2019-05-19:
For all those getting stuck at `MQTT.MessageFlag = true` when using software serial, ESP8266 or other platforms, change `GSM_MQTT::processing` in `GSM_MQTT.cpp` to
```cpp 
void GSM_MQTT::processing(void)
{
 if (TCP_Flag == false)
  {
    MQTT_Flag = false;
    _tcpInit();
  }
  _ping();

  //**************new code*****************
  unsigned long t = millis();
  //add long waitPeriod = 0; somewhere at the beginning of the file
  if (t - waitPeriod > 1000L) 
  {
    waitPeriod = t;
    serialEvent(); //call serialEvent every second
  }
  //*****************************************

}

closed 2019-12-17 by elementzonline

MuhammedImdaad, 2021-03-03:
code works perfectly fine!!!!!!!!!!!!!!!!!!