elementzonline/SIM800_MQTT #10: Using library with itead GBOARD

guillebot, 2016-10-07:
Hi:

I have a GBOARD, which is a arduino+sim900 integrated in the same card.

It works ok with other libraries, but I would love to use this because of its simplicity.

However, I suppose that because of different pin numbers and connections, I'm not getting it to connect.

I see the log via Serial Monitor, but it hangs on "+++AT".

Where should I look in order to change the serial port for communication between the arduino and the sim900 module?

Thanks in advance

Guillermo

guillebot, 2016-10-07:
To add a little more info, the library that it is working is using SoftwareSerial on pins 2 and 3 for Arduino-GSM communication, and hardware serial for logging.

guillebot, 2016-10-07:
Self response. It works perfectly with ITEAD GBOARD SIM900.

It was necessary to swap hardware and software UARTs, and define SoftwareSerial inside GSM_MQTT.cpp.

Thanks for a great library.

closed 2016-10-07 by guillebot

rociochueca2, 2019-01-14:
Hello Guillermo I try to connect the Gboard with a Ftdi connector, however can not. Please, let me know how to make this connection.

guillebot, 2019-01-16:
Hi there. I did this a long time ago, but it went just fine following the pinouts in:

https://www.itead.cc/wiki/Gboard_800  (yes, 800. It's the same)

You have to use a ftdi adapter like cp2102 and connect to the block of 7 pins that is under the XBEE PRO marking.