elementzonline/SIM800_MQTT #15: Stuck at +++AT output. [Using SIM808]

karanmakharia, 2017-01-04:
I am using arduino nano and SIM808. I tired the publish example but the serial monitor output is stucked at +++AT. Please tell me the solution. Also tell me how to attach SIM808 to software serial instead of default serial pins.

bhimavin, 2017-01-08:
I am also facing same issue for SIM808 and SIM800C looks like commands are different for these modules.
I found this link
http://arduino.stackexchange.com/questions/26670/sim808-module-gsm-working-fine-gps-not-working-no-gps-ready-message-on-power-on

nithinkurian, 2017-01-09:
Stuck at +++AT means, your device (arduino) is not getting OK reply from your modem so please,
           1. please send AT command directly to your modem and check whether it is giving OK reply or not
                    if it is not proper, rectify that issue first.
           2. if your modem is replying ok, then cross check, modem and device (arduino) connections

bhimavin, 2017-01-14:
Hello nithinkurian,
Thanks for reply.
I am still facing issue:
Serial.available() always 0 in void serialEvent() function, don't understand, why?

Tried this [example](https://github.com/chandankumar1/temperature_logger/blob/master/thingspeak.ino) getting response for all AT commands.

Verified serial connection with below code and got response for all AT commands
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

void setup()  
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  mySerial.begin(9600);

}

void loop() // run over and over
{
  if (mySerial.available())
    Serial.write(mySerial.read());
    
  if (Serial.available())
  { 
    while(Serial.available())
    {
      mySerial.write(Serial.read());
    }
    mySerial.println();
  }
}`

bhimavin, 2017-01-15:
Found solution for me, I was not using hardware serial for modem.
```
void setup()
{
  // initialize mqtt:
  // GSM modem should be connected to Harware Serial
```

Just need to correct connections.

nithinkurian, 2017-01-15:
Ok, so the issue is resolved

samawi, 2017-01-29:
I think I am also using hardware serial (which I believe are the ones with TX1 and RX0 labels) but still stuck at +++AT 

Need help please

siddharthtaunk, 2017-04-13:
comment out the '+++' serial print statement from the library and then it works fine.
Siddharth

Akelectro, 2017-04-30:
Hi Guys,
I am facing the same issue. 
1. Commented out '+++'
2. Using hardware serial to connect with SIM800 modem.
Still stuck at AT. Not getting OK command from the modem. 
I checked the modem by directly passing AT command and it is working. 
Any help is appreciated. Thanks

livzz, 2017-05-12:
If you are using SoftwareSerial interface with the GSM MODULE replace all `mySerial` with `Serial` at 

> GSM_MQTT.cpp
except for the `extern SoftwareSerial mySerial;` at line no 13

nedimhx, 2018-04-17:
@livzz could you please post somewhere full working code, I can get it working on arduino uno?

regards

closed 2019-09-14 by elementzonline

piter1235301, 2020-06-13:
Hello,
@livzz can you send this code with modification? I try edit this but this not help - maybe i made mistake...
Best regards