elementzonline/SIM800_MQTT #16: millis() issue?

durvesh09, 2017-01-13:
How to published message after specified time interval using millis() ?
facing problem in this

elementzonline, 2017-01-13:
Hi, Please refer the blink without delay sample code to use millis function for tasks. You can check if the MQTT connection is available before publishing.

closed 2017-01-13 by elementzonline

durvesh09, 2017-01-15:
Hi,

   Thank you for the help.I tried the same sample code, but its only
printing the publish message on serial monitor after specified interval but
not publishing it. Any solution ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

durvesh09, 2017-01-16:
Do i need to change the keepalive interval also for publishing messages
after specified time ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech