elementzonline/SIM800_MQTT #17: [Feature Request] Add SSL/TLS support

openbufo, 2017-03-21:
What if MQTT broker accepts SSL/TLS connection? In IoT, security stands first and SSL support is inevitable.

tusharvb19, 2017-09-05:
Need SSL/TLS support.
Isn't it easy by doing AT+HTTPSSL=1???

jameshowe, 2018-10-16:
@tusharvb19 old thread, however, given the library establishes a TCP then it would be `AT+CIPSSL=1`

rudolfbyker, 2018-11-15:
When working with AWS IoT, it seems that there are some extra requirements:

> Clients wishing to connect using MQTT with X.509 Client Certificate authentication on port 443 must implement the Application Layer Protocol Negotiation (ALPN) TLS extension and pass x-amzn-mqtt-ca as the ProtocolName in the ProtocolNameList.

Source: https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html

I'm not sure if that applies to other brokers, too.

elementzonline, 2018-11-16:
Hi,

I am not maintaining the repo anymore. Feel free to clone the repo and make
changes.

We are now using A9G module for Mqtt based communication. It supports SSL
communication as well.

krpagar, 2019-02-20:
Hello Dhanish,

Thanks for making library available for us !

I started with the library examples. I am able to connect to Adaruits.io
but not able to publish data continuously... In fact i don't know where to
and how add publish routine.. if could any sample simple code which sending
data to broker then it would be very helpful for me.

I hope i will learn the MQTT by knowing above things..

Expecting your favourable reply.

Thanks & Regards,
Kartik Pagar

elementzonline, 2019-12-17:
Closing due to inactivity

closed 2019-12-17 by elementzonline