elementzonline/SIM800_MQTT #22: Subsribing to the topic is not working

priya2212, 2017-06-20:
Hi everyone,
I am trying to work with this library to provide MQTT to SIM800 module. I am using Arduino UNO board. I am successfully able to publish the message on the topic but when I am subscribing to the topic and in the callback function, I am controlling the LED based on the message received:
**Subscription function:**
> void GSM_MQTT::OnConnect(void)
> {
>   subscribe(0, "qewhrquwkfh", "GPRSLED", 1);;
> }
Callback function:

> void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength)
> {
>   mySerial.println(TopicLength);
>   mySerial.println(Topic);
>   mySerial.println(MessageLength);
>   mySerial.println(Message);
> 
>   if (message[0] == '1')
>   {
>     digitalWrite(2, HIGH);
>   

  }
}
it seems like the callback function is not getting executed. It would be great for me, if anyone can guide me to resolve this issue.
Please help me!
Thanks & Regards,
Priya

arminsalcin, 2017-08-13:
Hello you need to subscribe in void loop exactly in " if (MQTT.available()) " function  and add line:

 MQTT.subscribe(0, 122212, "GPRSLED", 1);

elementzonline, 2017-08-14:
Issue due to incorrect usage of library. Please use the examples

closed 2017-08-14 by elementzonline