elementzonline/SIM800_MQTT #23: Working with Sim 808

himanshu-hearthacker, 2017-06-30:
Hello
I am using this library with sim808 (GSM+GPS), I am able to send data when i send any constant string but when i want to send GPS data, library doesn't publish GPS data. It probably disconnects TCP connection on running (CGNSINF=0) command. I tried to make TCP connection again but it doesn't work. Sometimes i am unable to get the GPS data when TCP connection enables. Is there any way i am able to send GPS data when TCP connection alive, I don't want to disconnect TCP connection to get GPS info and then again re-establishes TCP connection again. to send GPS data.

architmuchhal12, 2017-07-27:
I am also working with SIM808 to get an MQTT connection. Could you share your code? 
I have tried many different libraries but with no success. I was able to get to work the Adafruit MQTT (
[(io.adafruit.com)](url) but I want to use a different MQTT broker.
Thanks!

nedimhx, 2018-04-17:
@architmuchhal12 did you ever get it working, If you did, could you please send me full .ino project ?

regards

AhmadSalmanKhan, 2019-03-20:
Hi guys. Any update on this issue?
I have made a working prototype but there is a problem, it works fine in the start( for some time) and even publishes the location to mqtt broker but then disconnects and rarely connects back. I am getting socket disconnect error.
-using pubsubclient and TinyGSM libraries.

elementzonline, 2019-09-14:
The library is working in Transparent mode. So disconnecting and reconnecting GPRS frequently is not advisable.

closed 2019-09-14 by elementzonline