elementzonline/SIM800_MQTT #24: AutoConnect bug..

xoxoj, 2017-08-04:
why stop just can see the print AutoConnect, but can't go on..

this is my debug log...

`76
Disconnecting
0
CLOSED
at GSM_ReplyFlag !=燾oming begin...
AT

OK
+++
AT

OK
ATE1

OK
AT+CREG?

+CREG: 0,1

OK
AT+CIPMUX=0

OK
AT+CIPMODE=1

OK
AT+CGATT?

+CGATT: 1

OK
 modemStatus = 3 
 _tcpStatus = 2 
at GSM_ReplyFlag != 7
AT+CIPSTATUS

OK

STATE: IP INITIAL
else           _tcpStatusPrev == _tcpStatus
mySerial.print(_tcpStatus);   is ->>>> :
 <<<<<<<<<<<------
AT+CSTT="CMNET"

OK
at GSM_ReplyFlag != 7
AT+CIPSTATUS

OK

STATE: IP START
else           _tcpStatusPrev == _tcpStatus
mySerial.print(_tcpStatus);   is ->>>> :
 <<<<<<<<<<<------
AT+CIICR

OK
at GSM_ReplyFlag != 7
AT+CIPSTATUS

OK

STATE: IP GPRSACT
else           _tcpStatusPrev == _tcpStatus
mySerial.print(_tcpStatus);   is ->>>> :
 <<<<<<<<<<<------
AT+CIFSR

10.40.77.11
at GSM_ReplyFlag != 7
AT+CIPSTATUS

OK

STATE: IP STATUS
else           _tcpStatusPrev == _tcpStatus
mySerial.print(_tcpStatus);   is ->>>> :
 <<<<<<<<<<<------
AT+CIPSTART="TCP","test.mosquitto.org","1883"

OK

CONNECT
MQTT.TCP_Flag = True
AutoConnect`

arminsalcin, 2017-08-13:
Can you say us what gprs model you use what microcontroller ?

elementzonline, 2019-09-14:
Closing this as no reply given

closed 2019-09-14 by elementzonline