elementzonline/SIM800_MQTT #25: [QUESTION] ESP8266 compatibility

Franciscopcosta, 2017-10-20:
How can i adapt it to work with ESP8266?

kristofferJr, 2018-01-29:
I'm asking the same question

elementzonline, 2018-01-29:
Yes, there are some changes to be gone in AT commands, otherwise should be working fine. But writing code directly to support MQTT in ESP8266 is the straightforward approach.

agozie, 2019-05-19:
For all those getting stuck at `MQTT.MessageFlag = true` when using software serial, ESP8266 or other platforms, change `GSM_MQTT::processing` in `GSM_MQTT.cpp` to
```cpp 
void GSM_MQTT::processing(void)
{
 if (TCP_Flag == false)
  {
    MQTT_Flag = false;
    _tcpInit();
  }
  _ping();

  //**************new code*****************
  unsigned long t = millis();
  //add long waitPeriod = 0; somewhere at the beginning of the file
  if (t - waitPeriod > 1000L) 
  {
    waitPeriod = t;
    serialEvent(); //call serialEvent every second
  }
  //*****************************************

}

elementzonline, 2019-09-14:
Closing the issue as ESP8266 is offtopic and not intended as the target for this library

closed 2019-09-14 by elementzonline