elementzonline/SIM800_MQTT #28: Stuck at MQTT.TCP_Flag = True ( Sim 800 with atmega328)

gopi1230, 2018-05-28:
I am using Sim800 with atmega328. I tried subscribe example , first i struck at +++AT output . After i read issues , i change Serial and mySerial vice versa. i got below response`AT

+CGATT: 1

OK

OK
AT
ATE1

OK
AT+CREG?

OK

+CREG: 0,1

OK
AT+CIPMUX=0
AT+CIPMODE=1

OK
AT+CGATT?

OK
AT+CGATT=1

+CGATT: 1

OK
AT+CIPSTATUS

OK

OK

STATE: IP INITIAL
AT+CSTT="AIRTELGPRS.COM"

OK
AT+CIPSTATUS

OK

STATE: IP START
AT+CIICR

OK
AT+CIPSTATUS

OK

STATE: IP GPRSACT
AT+CIFSR

100.64.189.225
AT+CIPSTATUS

OK

STATE: IP STATUS
AT+CIPSTART="TCP","52.205.105.7","5555"

OK

CONNECT
MQTT.TCP_Flag = True`

Then after i changed serialEvent to serialEvent1 and serialEvent2 also , but same response coming now . Here , I am sending my code and circuit diagram.

[... 935 lines cut ...]
          else if (ReceivedMessageType == PINGREQ)
          {
            MQTT.TCP_Flag = false;
            MQTT.pingFlag = false;
            Serial.println("Disconnecting");
            MQTT.sendATreply("AT+CIPSHUT\r\n", ".", 4000) ;
            MQTT.modemStatus = 0;
          }
        }
      }
      else if ((inChar = 13) || (inChar == 10))
      {
      }
      else
      {
        Serial.print("Received :Unknown Message Type :");
        Serial.println(inChar);
      }
    }
  }
}`

[image]

Could you please help me out , i am trying it from one week.

Franciscopcosta, 2018-09-20:
Did you solved this?

krpagar, 2019-02-20:
Thanks for making library available for us !

I started with the library examples. I am able to connect to Adaruits.io but not able to publish data continuously... In fact i don't know where to and how add publish routine.. if could any sample simple code which sending data to broker then it would be very helpful for me.

agozie, 2019-05-19:
For all those getting stuck at `MQTT.MessageFlag = true` when using software serial, ESP8266 or other platforms, change `GSM_MQTT::processing` in `GSM_MQTT.cpp` to
```cpp 
void GSM_MQTT::processing(void)
{
 if (TCP_Flag == false)
  {
    MQTT_Flag = false;
    _tcpInit();
  }
  _ping();

  //**************new code*****************
  unsigned long t = millis();
  //add long waitPeriod = 0; somewhere at the beginning of the file
  if (t - waitPeriod > 1000L) 
  {
    waitPeriod = t;
    serialEvent(); //call serialEvent every second
  }
  //*****************************************

}

closed 2019-09-14 by elementzonline