elementzonline/SIM800_MQTT #31: Can't publish and subscribe

Franciscopcosta, 2018-09-21:
I get the message:

AT+CREG?

+CREG: 0,5

OK
AT+CIPMUX=0

OK
AT+CIPMODE=1

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIPSTATUS

OK

STATE: IP INITIAL
AT+CSTT="wm"

OK
AT+CIPSTATUS

OK

STATE: IP START
AT+CIICR

OK
AT+CIPSTATUS

OK

STATE: IP GPRSACT
AT+CIFSR

xxx.xxx.x.x (ip address)
AT+CIPSTATUS

OK

STATE: IP STATUS
AT+CIPSTART="TCP","_my_broker_ip_","1883"

OK

CONNECT
MQTT.TCP_Flag = True

But i cant publish and subscribe.
`if (MQTT.available())
  {
    Serial.println("MQTT available...");
    delay(1000);
    /*
      if you want to do something when mqtt connection is live.
      You can write your code here
    */
  }`
also this is not working, i never get the message "MQTT available..." on my serial.
Please can anyone help me?
(SIM800L with ESP32 via HardwareSerial UART2)

fdeizaguirre, 2019-02-28:
Dear @Franciscopcosta I'm having the same problem, did   you made any improvement?

marcosdepo, 2019-11-15:
Serial pins Rx and Tx work with SIM800 module.
This library uses software serial "mySerial"
Change Serial.println("MQTT available..."); with mySerial.println("MQTT available...");
Conect an external usb/serial driver, it works for me.

closed 2019-12-17 by elementzonline