elementzonline/SIM800_MQTT #38: +++AT loop, SIM800L, Arduino UNO

piter1235301, 2020-06-13:
Hello,
I have a problem connecting because I get +++ AT all the time. I saw a similar problem on this page: 
_https://github.com/elementzonline/SIM800_MQTT/issues/15_
but it doesn't really help.

1.  I tried this program: _https://github.com/chandankumar1/temperature_logger/blob/master/thingspeak.ino_ ,and all commend give me feedback OK - so i thing SIM800L and Arduino UNO it's ok.

2. I tried connect SIM800L to SERIAL PIN (0, 1) and SOFTWARE PIN (10, 11) - with no result.

3. @livzz  said: 

> If you are using SoftwareSerial interface with the GSM MODULE replace all mySerial with Serial and vice versa 
>     GSM_MQTT.cpp
>     except for the extern SoftwareSerial mySerial; at line no 13
> 

I tried this also, but same result... but maybe i made some mistake in code? I would be grateful if you could send the correct code...

Anyway i tried this with another module SIM800L and with Arduino nano... Same situation... 

What i can do next?

Best regards.

MuhammedImdaad, 2021-03-03:
> Hello,
> I have a problem connecting because I get +++ AT all the time. I saw a similar problem on this page:
> _#15_
> but it doesn't really help.
>
> 1. I tried this program: [...]

did you solve this. i am also facing a similar error

piter1235301, 2021-03-03:
Nope :(

MuhammedImdaad, 2021-03-03:
> […](#)

easy. use software serial

tcampos123, 2021-03-25:
I have the same problem, how can I solve it?

MuhammedImdaad, 2021-03-26:
either use software serial. or don't interrupt the serial line mainly the TX line.

closed 2021-07-15 by elementzonline